cooked/kimotor #31: Can't install

rozrabiak, 2023-01-10:
Kicad 6.0.10

[image]

cooked, 2023-01-10:
Hi @rozrabiak ,can you provide:
- which OS are you on?
- where Kicad was installed from (download from homepage, repo or flatpak if on Linux)

thanks, that would help to troubleshoot the error

rozrabiak, 2023-01-10:
Yeah, sorry,
Windows 11 x64.
Kicad installed from homepage.

cooked, 2023-01-19:
Hi @rozrabiak ,
thanks for the info on your system. Additionally it would help me if you can indicate the steps that got you to the error ( for example if you changed parameters before generating or if you ran the default settings).

Thanks again

rozrabiak, 2023-01-19:
I'm downloaded as zip file from GitHub and install zip file.
I didn't change any parameters.
When I try installing from kicad via download option - there was a file download error.

cooked, 2023-01-22:
Hi @rozrabiak 
v0.1.1 should be available in the Kicad Content Managare tomorrow Jan 23 at 12pm.

It contains a fix for issue #30 that others are experiencing, but possibly will fix also yours.  

If you test it out let me know how it goes

cooked, 2023-03-18:
@rozrabiak is this solved for you?

cooked, 2023-05-21:
in between official releases, that will be available from KiCad PCM, the plugin can be installed as described here: https://github.com/cooked/kimotor/issues/39#issuecomment-1550412060

closed 2023-05-21 by cooked