cooked/kimotor #38: default parameters generate bad layout (Win, Kicad 6.0.10)

cooked, 2023-01-25:
It looks like there are other issues.  I ask KiMotor to "Generate" from default values which show in the dialog and get an image which looks like it has PCB drawing issues using KiCad v6.0.10.  I should probably open a new issue for this input.

[image]

rozrabiak, 2023-02-12:
That's true..
Kicad 6.0.11

[image]

cooked, 2023-02-12:
Hi @rozrabiak,
thanks for confirming. I also did run a test and so far it looks like I could reproduce the bug on Window as well, while Linux works ok.
I'll have to look into the algorithm that generates the coil layout which is the one causing this issue.

Hope to get back to you with a positive update

HaydenHu, 2023-04-30:
> @rozrabiak as a quick workaround you can try to disable the track fillet, that is the part that is causing the issue... [image]
> 
> with the track fillet set to 0 you should be able to have a proper layout, until I've validated the fix to the fillet algorithm that I'm currently working on
> 
> let me know how it goes
> 
> P.S. if you are testing on kicad 7, you will still not be able to run the plugin at the moment, but I plan to add the support for the 7 this weekend

Have you solved the rounded corner algorithm?

cooked, 2023-05-21:
> Have you solved the rounded corner algorithm?

Hi @HaydenHu, unfortunately the issue is not solved. Since the bug appeared on Win, the plugin code from the repo now disables the fillet on Win OS, to prevent triggering this bug

cooked, 2023-10-28:
fixed in 0a86122faedac84f7919117dfa3418df440ff211
don't shout it too loud!

closed 2023-10-28 by cooked