cooked/kimotor #39: Kicad 7.0 - no icon for kimotor

rozrabiak, 2023-02-13:
In kicad 7.0 (today release) after install kimotor no icon exists on toolbar, can't use kimotor.. :-(

Installed from plugin manager.
Windows 11.

HaydenHu, 2023-03-09:
It seems that "pcbnew.IU_PER_MM" is wrong

cooked, 2023-03-18:
Hi @rozrabiak @HaydenHu,
I've just added kicad 7 support (commit https://github.com/cooked/kimotor/commit/c045b4b2e7f5446c20664362f29c8b77610b62f4). This is likely  the reason why you didn't see the icon before.

However, I'm still working on more bug fixes and I'm not ready to push the new version to the plugin manager yet.
To try it out you can download the zipped master branch of this repo and import tha zip in kicad.

[image]

[image]

Let me know if that works for you.

lukebayes, 2023-05-11:
Really excited to try this out, thanks for sharing!

I'm running on Kicad 7 (also tried in Kicad 6) on both Windows and Linux.

Installing from the file as described fails with the following error: "Archive does not contain a valid metadata.json file."

I did some quick checking and found [this conversation](https://forum.kicad.info/t/metadata-json-archive-missing/38753) about this error.

lukebayes, 2023-05-11:
I was able to get unblocked by unpacking the downloaded zip and found that Github nested the folder. The missing metadata.json file is inside an inner folder. I zipped up that folder's contents and got one step further. The installation appeared to succeed.

Still not seeing an icon anywhere, but at least it's not throwing an error. Thanks!

katbinaris, 2023-05-12:
Hi! I can confirm the same issue:
OS1: Win
OS2: Ubuntu
KiCad version: 7.0
KiMotor Install 1: Manual from master branch 
KiMotor Install 2: Plugin Manager
KiMotor version: master

Plugin installs however icon is nowhere to see. So it is not in PCBNew Action panel (nor is in preferences)
I'm trying to troubleshoot it and see whether kimotor is not being initialized, or whether i have issue with wx, though other plugins that have wx dependency works fine. Also seem not to be an OS issue as i can replicate it on both Linux and Win11

katbinaris, 2023-05-12:
It does seem to be wx.libpubsub deprication issue. 
Will try to instal PyPubSub and see if that makes a difference.

cooked, 2023-05-15:
> Really excited to try this out, thanks for sharing!
>
> I'm running on Kicad 7 (also tried in Kicad 6) on both Windows and Linux.
>
> Installing from the file as described fails with the following error: [...]

Hi @lukebayes thanks for reporting, I'm still trying to iron out issues arising from moving to kc7.
However, for the time being, with Kicad6 on Linux  the plugin is available scrolling down in the Plugin and Content Manager (i.e. from the official repo and not from local file).

hope that works for you in the meantime

cheers
stef

cooked, 2023-05-16:
For kicad 7, trying to install the latest KiMotor, what I've tested working at the moment is:

1) download the KiMotor archive from the repo as described above https://github.com/cooked/kimotor/issues/39#issuecomment-1474757452 but, instead of using the PCM to install the addon, just unpack it manually to:
- C:\Users\<youruser>\Documents\KiCad\7.0\3rdparty\plugins  (Windows)
- /home/<youruser>/.var/app/org.kicad.KiCad/data/kicad/7.0/3rdparty/plugins/ (Linux, Kicad installed from flatpak)

2) alternatively (to the .zip) do "git clone https://github.com/cooked/kimotor.git" in one of the folder described, according to your OS  (or in any of the folder Kicad will look for addons https://dev-docs.kicad.org/en/python/pcbnew/)

after that restart Kicad. the plugin icon should be there in the toolbar.

closed 2023-05-27 by cooked