cooked/kimotor #45: BUG: package not working

hstarmans, 2023-09-20:
I checked this package. There is a version which works for Kicad 7 shared on https://forum.kicad.info/t/kimotor-plugin-fix/40836/1 .

I was able to get some results.
The coils are nicely packed but:
 -  I doubt the Kicad motor actually works, for four layers the resulting motor is not working
 -  On windows there are issues which track generatio

I think the package needs a whole rewrite and should be tested by "building an actual motor".
For the time being i am using kicad-coil-plugin from atomic 14 https://github.com/atomic14/kicad-coil-plugins/

cooked, 2023-09-22:
Hi @hstarmans ,
appreciate the feedback. Can you provide some context to the issue? it would help to troubleshoot the problems you're experiencing if you can provide:
- for the motor that doesn't work, a screenshot of the settings you used to generate it (on top of the 4lyrs you already mentioned), one of the PCB in kicad and one of the produced PCB... additionally would help to know what was the config of the magnets on the rotor (nr and arrangement)
- about the broken tracks and vias not connected/located correctly also the settings that generated them and a screenshot would go a long way
- the bug regarding track generation on windows is tracked in issue #38, although not fixed yet, setting fillet radius to 0 on win should generate a proper layout, but let me know if that's not the case for you

hstarmans, 2023-09-25:
**Settings**:
I adapted your program so I could enter these settings.
[image]

**Issues**

*Overlapping / broken tracks*
[image]

*4 layers*
All layers connect to this point. 
[image]
In a normal electromagnet all coils are in series. In your boards, part of the coils are in parallel. This increases current.
Magnetic strength is linear with current. Heat dissipation goes via the square.
As such, your board might not be optimal if heat is the limiting factor.
Furthermore, in your board in2 goes against front copper. As a result, it does not work.

*Weird start*
The start here is weird. It first goes left and then right. This is not ideal.
[image]

Summarizing, I decided to use  https://github.com/atomic14/kicad-coil-plugins.
My motor is shared here https://github.com/hstarmans/firestarter/tree/master/lasermodule/pcbmotor_spiraloptimized 

What I liked about your plugin;
  - does not require notebook, works directly from Kicad
  - calculates resistance

Still, atomic is able to provide systems with more layers and 12 coils.
He also provides an approximation of the electromagnetic field and did build a motor.
If you want to go further with this, I would look at his work. 
Also look at the notes in my mark down here https://github.com/hstarmans/pcb-motor
Especially this link https://things-in-motion.blogspot.com/2019/01/selecting-best-pole-and-slot.html

cooked, 2023-09-25:
Hey @hstarmans ,
thanks a lot for taking the time and sharing, you gave me really valuable info. I'll make a list of GitHub issues tonight, to plan the work a bit. I'll start removing hard limits on the input fields, which is a common request, and try to reproduce your layout and possibly adjust the code to fix that first.

hstarmans, 2023-09-28:
@cooked 
I would also setup [pre-commit](https://www.architecture-performance.fr/ap_blog/some-pre-commit-git-hooks-for-python/). Here you see an example of a [config file](https://github.com/sktime/sktime/blob/main/.pre-commit-config.yaml). You can auto format your code with black and ruff.

Once you " finished"  your results you can automate the checks with  https://github.com/INTI-CMNB/KiBot
Furthermore, try to keep it minimal, focus on one thing and do it well.

cooked, 2023-10-15:
@hstarmans over the weekend I finally managed to spare few hours for coding. the elephant in the room was the algo that would handle the "wiring" of high slots count motors.
I've run it only on Linux so far, but with the new code, also the coil generation seems to have improved a bit.

any feedback is always appreciated :-)

I put here sample for 15 and 18 "slots" on 6-layers PCB...still quite large outline motors compared to your case, but next I will test smaller ones
[image]

[image]

here the details of the terminals, poles connection segments and star connection on the inner rings (inner layers hidden as all the connections happen on F and B)

[image]
 

One thing to note is that for layers >2 the coils jump across layers with buried vias...you can check that navigating to the inner layers and right click on the via

[image]

weeenerplastics, 2023-10-17:
Apologies, replying from a different account.
The package improved a lot.
On windows 10, my setttings look slightly truncated.
[image]
The coils look indeed much better. 
I checked the four layer board and cannot see a lot of mistakes. I do think the red line above the via 1-2 should be removed in the picture shown below.  It also shows up in the DRC checks, with a warning unconnected end.

[image]

I have more issues in my Design Rules Checker.
I got a couple (Isolated copper fill, annular width board setup constrains 0,05, actual -0.0750). This might be something too look at.
Anyhow, great work!

cooked, 2023-10-22:
Hi @hstarmans ,
you might want to check the latest commits, I've been doing some work to fix the via issue + other fixes for the DRC sake.
Still some works remain for co-located vias (on 4+ layers) for which I have to implement some staggering algorithm.

regarding the compressed GUI on Win, is this maybe your situation https://github.com/cooked/kimotor/issues/37 ? I do have tested the plugin on Win10 (1920x1080) here and it shows just fine

cooked, 2023-10-28:
closing, since most is taken care of.
I've moved your suggestions/improvements to new issues
https://github.com/cooked/kimotor/issues/54 
https://github.com/cooked/kimotor/issues/55
https://github.com/cooked/kimotor/issues/51
and added the problem with the GUI in
https://github.com/cooked/kimotor/issues/37

closed 2023-10-28 by cooked