cooked/kimotor #46: Idea : Possibly combine the below to the current version

geekx, 2023-09-24:
https://github.com/atomic14/kicad-coil-plugins

In this repo, it has a wedge version coil generator, would also help

cooked, 2023-09-24:
Hi @geekx,
the development version (the code in this repo master branch) got the option to pick "radial" as a strategy for coil layout. That should generate the wedges you're looking for, if I understand your question correctly.

[image]

radial
[image]

parallel
[image]

cooked, 2023-10-22:
Hey @geekx, 
yes I would advise to still use the DRC. At the current stage the plugin does not enforce any fab constraint, so having proper board settings in the project + taking care of DRC messages, are still the steps that guarantee that you meet your PCB fab capabilities.

closed 2023-10-25 by cooked