cooked/kimotor #49: KiCad 7.0 Support

Pana1v, 2023-10-15:
Can you please give me steps to import this plugin into Kicad 7.0. Having trouble with it.

Pana1v, 2023-10-15:
I've installed it using the plugin manager but it still isn't working

cooked, 2023-10-19:
Hi @Pana1v
I've updated the README page with some installation guidance.
Can you have a look and see if that  helps you?
I suggest to uninstall the one from the package manager and either clone this repo or download it as a zip file.

Then unpack it in the folder described there.
Let me know if you need further help or if it just works 
cheers

https://github.com/cooked/kimotor/blob/master/README.md

ostapio, 2023-10-19:
@cooked probably make sense also to include information on how to install numpy for kickad provided python :) that what i had to do on macos clean 7.0 installation, also there r few places where code brakes so i needed to fix it in order to work correctly, im planning to submit PR at some point with fixes (not sure if its only macos related but i dont have linux box right now to check)

Pana1v, 2023-10-20:
The new documentation and features look great, unfortunately the older one was working in KiCAD 6.0.7. Now it isn't working in either. I click generate and nothing appears. In a few cases the count of tracks and vias did update but nothing was visible. Thanks in advance.
[image]

cooked, 2023-10-22:
Hi @Pana1v, 
I just fixed a typo (in one of the footprint internal db) that was preventing the layout generation to complete.
I've tested it on my Mac and seems to work.
Let me know if that works for you as well.

cheers

Pana1v, 2023-10-24:
Hi, here's the backtrace output

`Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
`

Pana1v, 2023-10-24:
Hi, here's the backtrace output

Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range
Traceback (most recent call last):
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 948, in on_btn_generate
    self.generate()
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 83, in generate
    self.do_junctions( coil_t, ext_t, int_t)
  File "Z:\Documents\KiCad\6.0\3rdparty\plugins\com_github_cooked_kimotor\kimotor_action.py", line 528, in do_junctions
    j.SetEnd( int_t[3][p] )
IndexError: list index out of range

Pana1v, 2023-10-24:
Hi, so the Kicad 7 gives and error, no metadata.json

cooked, 2023-10-28:
fixed in 9a43304de56537820a9490748ec31b7c1365e530
added support for EDA_ANGLE and provided installation instructions to make sure numpy is installed, in particular on Mac OS)

closed 2023-10-28 by cooked