cooked/kimotor #51: co-located vias shall be avoided

cooked, 2023-10-22:
on motors with 4+ layers, the buried vias used to jump between layer pairs are now "co-located" (same x,y coordinates)

this is flagged as a warning by the DRC since PCB fabs might not be happy with this setting, and the tolerances on drilling a hole in the same spot multiple times might lead to failing PCBs.

to avoid this the algorithm should "stagger" the buried vias along the arc track in a way that the min hole-hole clearance is satisfied

[image]

cooked, 2023-11-01:
looking into PCB fabs capabilities it is clear that to keep the cost of low-volume boards low, blind/buried vias shall be avoided. a possible approach is to move via away from coil tracks so that regardless of the nr of layers the layer jumps can be achieved with through holes (standard vias)

[image]

algocrypto, 2024-03-28:
Hi,
Thank you for sharing the plugin, it looks to good for designing PCB stator. 

I am trying to build the PCB stator and am getting lot of errors, Is there any fix for this issue? 
[image]

cooked, 2024-04-29:
closing. follow up in related issue #61

closed 2024-04-29 by cooked