cooked/kimotor #53: New Coil Shape : Pentagonal

Pana1v, 2023-10-24:
@cooked I want to help add a new shape for coils - Pentagonal or star shape.
Can you guide me on this. I'd like to learn and contribute to the project.

cooked, 2023-10-25:
Hi @Pana1v,
thanks for the support. Possibly the pentagonal shape is similar to what I had implemented originally, but then discontinued.

Just out of curiosity, is the shape you need dictated by mechanical constraints or some EM design considerations?

Anyway, to kickstart I would have a look at the steps involved in the generation of the layout, they're called in generate() and they all named **self.do_<something>()** https://github.com/cooked/kimotor/blob/master/kimotor_action.py#L205 ...the names should be mostly intuitive but for nomenclature sake:
- **coils** are for me just the loops on various layers
- **races** are the arc track segments placed inboard wrt to the coils
- **junctions** are the tracks connecting coils and races. 

The creation of the coils happens in **do_coils()** and it's also a 2-step process: 
- generate the coordinates of the "corner" points (as well as mid points for arcs), that's the job of **ksolver.radial()**
- connecting the dots to make the tracks and add them to the board, done by **coil_tracker()**

Do you have a sketch of the layout that you want to create? that would help the discussion

cheers
stef

Pana1v, 2023-10-26:
Hi, before finding KiMotor i worked with atomic14's plugin which could generate limited but 2 variants of this wedge shape 
[image]
Was trying if I could learn by trying to implement the same here.

cooked, 2023-10-27:
Hi @Pana1v,
ah I see. Not sure how atomic14 ended up with that specific shape. However, reading his blog that in the end the intuition that more copper is better, found confirmation also in the engineering approach with proper calculation of the magnetic field generated.
I believe that the maximum fill (and longest coil sides, the ones most important for the tangential component of B ).

Anyways, about the implementation, as a starter you can try to achieve a similar shape to atomic14 keeping the current radial solver but making your own version of the "tracker" that will use just straight line instead of lines+arcs (see picture below, points are actually the same for the two shapes, just the tracks change).

[image]

Pana1v, 2023-11-02:
hi, the recent update gives me this. Is it intentional or an issue. I got no backtrace.
[image]

cooked, 2023-11-02:
Hi @Pana1v, 
outer coil diameter must be greater than inner. Then make sure the loops are not too many otherwise they will overlap.
[image]

Pana1v, 2023-11-03:
Sorry for that silly error, I'll get started on pentagon. Thanks

cooked, 2023-11-06:
> I'm learning from scratch here, So I replaced PCB_ARC with PCB_TRACK, now I have flat ends, just need to add a mid point. So, I need to understand how the corner points are stored in mdp could you help me out. [image]

Hi @Pana1v,
**mdp** is the sequence of points of the coil (ordered from external to internal of the "spiral") stored as numpy.matrix   

Over the weekend I've started refactoring to make the code more readable and nomenclature consistent (this work is in branch **new_vias** for now). 
Together with that I also started to write down some [developer's notes](https://www.stefanocottafavi.com/kimotor-developer-guide/ )
Those can help developers to understand how KiMotor does things at the moment. It briefly describes (according to the consolidated terminology) the role of the "coil_planner" in generating the waypoints (the current **mdp**) and the "coil_tracker" in connecting them. Bear in mind that is the "tracker" to give a meaning to the waypoints (e.g. corners or arc mid points) so that 3 points can be connected in fact as a single arc or 2 segments

Here the workflow that is not yet rendered in the notes:
[image]