cooked/kimotor #58: How to access KiMotor

person678, 2023-11-11:
I'm sure I'm going to feel like a fool that I've missed something obvious, but how on earth do I open the KiMotor menu / begin designing? I'm on KiCad 7.0 on MacOS, I've installed to the correct folder specified and the plugin is showing up in the content manager, but I'm not seeing anything on any menu bar or any button that related to KiMotor. Might be worth adding this to the readme?

person678, 2023-11-11:
I fixed the problem, which turned out to be more complex than I expected. Numpy was installed with the KiCad installation, but I am on a Apple Sillicon Mac and the installed numpy was compiled for x86. I had to uninstall numpy, and reinstall it (just the standard pip3 install command) which got the correct version and now it's working.

 Also worth noting that the pip3 executuable was in /Applications/KiCad/.../3.9/bin rather than just 3.9.

closed 2023-11-11 by person678

cooked, 2023-11-13:
Hey @person678 , that was quick :+1: 
thank you for sharing, I'll make a note in the plugin homepage (since I doubt KiCad will fix this anytime soon)