cooked/kimotor #59: Feature Suggestion - Increase line width as as radius increases

person678, 2023-11-12:
I'm currently performing research for my master's thesis which will involve building a PCB motor. I've come across a paper which illustrates the impact of unequal line width for radial windings. The gist of the idea is that there is a lot of wasted PCB area as the trace heads towards the outer radius, so you can widen the conductor as it approaches the outer edge to increase current density, etc. 
You can see what I mean here: 

[image]

I'm not sure if KiCad has support for tapering a line's width - I imagine you would probably need to split those lines into small segments and increase the width of each one gradually?

cooked, 2023-11-13:
Hi @person678,
I see your point. Can you share a reference to the paper?

person678, 2023-11-13:
[This is the first paper](https://ieeexplore.ieee.org/abstract/document/8896048 ). The results are a bit difficult to interpret but I believe this is the key graph: 
[image]

The context here is a PCB stator generator being ran at a given speed (x axis). The output power for a given speed appears to be significantly better for the tapered trace. There is another graph which suggests the eddy current loss is much higher for the tapered trace, which seems counter-intuitive, but the graph suggests this doesn't counteract performance gains? I do wonder if they got the copper loss figures back to front...

[This other paper](https://ietresearch.onlinelibrary.wiley.com/doi/full/10.1049/iet-epa.2020.0622) (which references the first) investigates tapered traces for a radial design. They show significantly reduced copper loss (17%!) and resistance, though lower inductance too. The ones to compare are "Parallel" and "UEW". 

[image]

The UEW (tapered trace) also displayed much lower temperature than the others. 
[image]