cooked/kimotor #61: Feature Suggestion - Through vias instead of blind/buried

Sleepyboi7973, 2024-03-01:
Blind/buried vias are a significant price increase compared to through vias, would be nice to have the option if possible

cooked, 2024-03-01:
totally agree. some work is in progress in the new_vias branch... currently organizing vias along the tg direction, but also radially would work
[image]

algocrypto, 2024-04-12:
Hi there, 
I am having the same issue from JLCPCB --> 
.It seems that there are buried or blind via in your file and we don’t make PCB that with buried or blind via. 
Is there any fix for these buried or blind via, can we fix it manually?

cooked, 2024-04-29:
Hi @algocrypto,
I merged the new vias into master. No more buried vias, although there might be issue with the annular ring width still (I plan to tackle that next, having the algorithm to pick-up track width and clearance from the board settings).

This is how it looks for a 8-layer widing (the inner layers might still have some cosmetics issues where the end of the coil is not properly trimmed down, but that should not impact functionality)

[image]
 
[image]

Let me know how it works for you.