cooked/kimotor #63: GUI - Mounting holes incorrect in output.

algocrypto, 2024-03-28:
Hello, 

It seems that the mounting holes paramater is not working on GUI, if I select 5 holes or any number of holes, the output is always 6 holes. 

Please refer to the attached screenshot. 

Am I missing something or does this look like a bug? 

Cheers!! 

[image]

cooked, 2024-03-29:
Hi @algocrypto,
have a look at the installation section of the [README file](https://github.com/cooked/kimotor#installation) and give it a try to one of the two "manual installation" methods. 
It seems you're on the plugin version from the package manager, which has now fallen behind, while the code from the repo contains the fixes you need.

Before installing the new version make sure to remove the one installed via Package Manager.

The update possibly solves also #64

algocrypto, 2024-03-30:
Hi @cooked ,
Thank you for the quick update. 
 I have installed the new version 0.1.2 manually as per the instructions. 
The problem still exist and I do not see any changes. Please refer to the attached screenshot for the PCB generated from the new version of the plugin. 
Cheers ! 
[image]

cooked, 2024-03-31:
mmm... it looks like kicad is still using the old version. the GUI in the new one is slightly different (see below) 

try to (and/or): 
- restart kicad
- refresh plugins (in PCB editor navigate to *Tools > External Plugins >  Refresh Plugins* ) 
- check in the package manager local folder if the old plugin is really gone
- make sure that you cloned the repo to the proper kicad version folder (in the README this is 7.0 but if you are on 8.0 that is where you have to clone the repo to)

PS: in the meantime I also pushed a new version of Kimotor  to the package manager... maybe you can give that I try first

[image]

algocrypto, 2024-03-31:
Hi @cooked ,

Thank you for the quick updates.

I have tried both manual and package manager installation, and I still getting the old GUI, please see the attached screeshot 

I have tried to restart KiCAD and reinstall, but same results. 

[image]

algocrypto, 2024-03-31:
Hi Stefano,

Thank you for the quick updates.

I have tried all the below recommendations, but I am still having the old GUI.

Please refer to the attached screenshots.

It looks like the GUI file on GitHub is not updated, it is still showing the last update was 5 months ago.

Kindly let me know if you need any more information to help fix these issues?

Once again thank you and best regards,
Sanjay

[Screenshot 2024-03-31 at 9.28.16 AM.png]

cooked, 2024-04-27:
Hi @algocrypto,
sorry for keeping you wait. Forget about the whole GUI thing, that was a problem on my end.

I've hopefully fixed the code to allow for any MH count when "Circle" outline is selected. This might bring up new issues, where you have to pay attention that the MH generated do not overlap with motor terminal (that was the reason to have them fixed to 6 before).

As a side note, in case you want to try other outline geometries, keep in mind that:
- if the nr of MH selected in the GUI is 0 no MH is generated
- if the nr of MH selected in the GUI is > 0, the nr of MH generated will be always equal to the nr of edges of the outline
Again, this is to avoid, outline + MH crazy scenarios (that shall be handled manually then)

Let me know if that works for you.

algocrypto, 2024-05-18:
Thank you @cooked for the update, the new GUI seems to be working for me now and I am able to get desired MH. 
But I am seeing other issues, for example, the via connector is always less, I tried to increase the track width expecting to have larger via holes. FYI, JLCPCB prefer to have 0.3/0.45 via holes else they have extra charges for PCB fabrication. I am still getting errors for via holes and many warning messages that the track is not connected. I think we are almost near the finish line for Kimotor plugin and I hope you will be able to fix it soon. 
Once again thank you for all the help.

cooked, 2024-06-09:
Hi @algocrypto
with the latest commit I've introduced a little change that hopefully will simplify and improve reliability of the workflow.
Based on PCB preset dropdown the following will happen if these are selected:
1. DRC rules: track width, clearances and via drill/hole (and annular width) are calculated based on the min allowed by DRC constraints
2. Current T/V: track and via settings used are the current ones from the lists in the kicad GUI. other parameters are derived and checked against DRC constraints.

Default workflow for me is to select **DRC rules** as Kimotor PCB preset  and in Kicad's **Board Setup > Design rules > Constraints** input all the geometrical parameters I want. If those match the JLCPCB requirements then the PCB motor should be compliant too.

Note that  there might still be warnings related to silkscreen. For the time being I suggest you to move text around manually, if needed.

looking fwd to your feedback

Katchyuser, 2024-06-09:
Awesome, Thank you @cooked , Let me check it out today and get back to you. I hope to complete the Stator design soon since it is delayed for long time and I am eager to test the new Axial Flux generator.

Katchyuser, 2024-06-10:
Thank you @cooked , The DRC - Annular width error is resolved for me..I have submitted the PCB stator design to JLCPCB for review, I will let you know if there is any other issue. I really Appreciate your help and support to the community.