cooked/kimotor #64: DRC - Annular width error

algocrypto, 2024-03-28:
Hi ,

I am generating the PCB and when doing DRC, I am getting around 300+ error for Annular width error for via holes. 

I am using the KiCAD version 8.0.1 if that helps to understand issue. 

Can you please have a look @cooked and help me to resolve this issue? 

Thanks,
[image]

algocrypto, 2024-03-30:
Hi @cooked ,
Thank you for the quick update. 
 I have installed the new version 0.1.2 manually as per the instructions. 
The problem still exist and I do not see any changes. Please refer to the attached screenshot for the PCB generated from the new version of the plugin. 
Cheers ! 
[image]

algocrypto, 2024-04-07:
Hi @cooked , Did you get a chance to look into the Annular width issue? I am still getting the error after using the updated version. Also I have downgraded to KiCAD 7.0 to make the plugin work, the new GUI was not working on Kicad 8.0

cooked, 2024-06-09:
hi @algocrypto see response in https://github.com/cooked/kimotor/issues/63 thatshould cover this issue as well