cooked/kimotor #71: Phase resistance and inductance showing 0

algocrypto, 2024-06-20:
Hello @cooked , 
I have received the PCB stator from JLCPCB today.
Upon checking the resistance between terminal pads on the PCB, I am getting resistance of 111 ohms. I assume this is too high resistance, and it will impact the output efficiency.
Is it possible to have the terminal resistance and inductance values in KiMotor plugin while designing? 

[image]

cooked, 2024-06-22:
@algocrypto yeah, let's fix that.
would you be able to share the kimotor details details of the pcb you produced (track width, clearance, nr layers, poles, etc..)? that would help me check the math against a known PCB motor model :-)

algocrypto, 2024-06-22:
Sure @cooked , Please find the attached parameters I have used and also the final PCB fabricated from JLCPCB.

As I mentioned,  the resistance between two terminal pad is 111 ohms which needs to be fixed.

Please let me know if you need any more information. 

[image]
[image]
[image]

algocrypto, 2025-01-13:
Hi @cooked Happy New Year 2025! Did you get a chance to look in this issue and help to fix it? TIA